mxtommy/Kip #644: Infinite loop starting KIP ?

gregw, 2025-05-26:
I've finally got my Calypso wind instrument wired into signalK, but it has sent KIP into and infinite loop, potentially from some bad data due to it thinking it has a compass but does not really.

Here is an example of a delta it sends:
```
{"updates": [{"$source": "calypso-up10", "values": [{"path": "environment.outside.temperature", "value": 273.15}, {"path": "environment.wind.angleApparent", "value": 2.670353755551324}, {"path": "environment.wind.speedApparent", "value": 5.6}, {"path": "navigation.attitude.roll", "value": -1.5707963267948966}, {"path": "navigation.attitude.pitch", "value": -1.5707963267948966}, {"path": "navigation.attitude.yaw", "value": 6.283185307179586}, {"path": "navigation.headingMagnetic", "value": 6.283185307179586}, {"path": "electrical.batteries.99.name", "value": "Calypso UP10"}, {"path": "electrical.batteries.99.location", "value": "Mast"}, {"path": "electrical.batteries.99.capacity.stateOfCharge", "value": 1.0}]}]}
```

which I can see in the databrowser fine, but when I try to start KIP I get this infinitely looping in my output and the window never loads and has to be killed.

```
POST /signalk/v1/applicationData/user/kip/11 500 1.792 ms - 203
PatchConflictError: Array location out of bounds or not an instance property
GET /@mxtommy/kip/assets/svg/icons.svg 304 1.090 ms - -
GET /@mxtommy/kip/ 304 2.516 ms - -
GET /@mxtommy/kip/chunk-EAJKXTFC.js 304 1.384 ms - -
GET /@mxtommy/kip/assets/tween-min.js 304 4.696 ms - -
GET /@mxtommy/kip/assets/steelseries-min.js 304 3.893 ms - -
GET /@mxtommy/kip/assets/NoSleep.min.js 304 3.331 ms - -
GET /@mxtommy/kip/chunk-ACMPUT6Z.js 304 2.815 ms - -
GET /@mxtommy/kip/chunk-OTD7HT32.js 304 2.290 ms - -
GET /@mxtommy/kip/chunk-AYIHSTCC.js 304 5.605 ms - -
GET /@mxtommy/kip/chunk-SHKAX4XV.js 304 6.579 ms - -
GET /@mxtommy/kip/chunk-JZUG3W7P.js 304 5.455 ms - -
GET /@mxtommy/kip/chunk-AB7BX4CL.js 304 4.217 ms - -
GET /@mxtommy/kip/chunk-MR7RUNKP.js 304 3.103 ms - -
GET /@mxtommy/kip/chunk-WE4P4WBD.js 304 2.243 ms - -
GET /@mxtommy/kip/chunk-36TH3DJS.js 304 1.482 ms - -
GET /@mxtommy/kip/chunk-QMIJ5LO6.js 304 4.305 ms - -
GET /@mxtommy/kip/polyfills-2DK6FBPF.js 304 3.591 ms - -
GET /@mxtommy/kip/main-QTKHCBQH.js 304 6.263 ms - -
GET /@mxtommy/kip/styles-ILFVF6BG.css 304 5.470 ms - -
GET /@mxtommy/kip/media/KFOmCnqEu92Fr1Mu4mxKKTU1Kg-SNGEW7FX.woff2 304 1.211 ms - -
GET /@mxtommy/kip/chunk-CD375FDG.js 304 3.543 ms - -
GET /@mxtommy/kip/chunk-SGYCFQGT.js 304 3.090 ms - -
GET /@mxtommy/kip/chunk-6QHCHGTT.js 304 4.782 ms - -
GET /@mxtommy/kip/chunk-EQAHQINY.js 304 2.716 ms - -
GET /@mxtommy/kip/chunk-7TWI6ZJI.js 304 2.229 ms - -
GET /@mxtommy/kip/chunk-EQ5CPWIU.js 304 1.755 ms - -
GET /@mxtommy/kip/chunk-Y7X77HSW.js 304 0.981 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOlCnqEu92Fr1MmSU5fBBc4AMP6lQ.woff2 304 4.133 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOlCnqEu92Fr1MmSU5fChc4AMP6lbBP.woff2 304 5.927 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOmCnqEu92Fr1Mu7GxKKTU1Kvnz.woff2 304 3.170 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOmCnqEu92Fr1Mu4mxKKTU1Kg.woff2 304 2.562 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOlCnqEu92Fr1MmEU9fChc4AMP6lbBP.woff2 304 2.037 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOlCnqEu92Fr1MmEU9fBBc4AMP6lQ.woff2 304 1.531 ms - -
GET /signalk/ 304 1.077 ms - -
GET /@mxtommy/kip/manifest.json 304 1.286 ms - -
POST /signalk/v1/auth/login 401 0.202 ms - 56
GET /@mxtommy/kip/assets/favicon.ico 304 2.894 ms - -
GET /@mxtommy/kip/assets/manifest-icon-192.maskable.png 304 2.217 ms - -
GET /@mxtommy/kip/chunk-WVCNHJ5A.js 304 3.207 ms - -
POST /signalk/v1/applicationData/user/kip/11/default 200 2.238 ms - 23
GET /@mxtommy/kip/assets/notification.mp3 304 1.266 ms - -
POST /signalk/v1/applicationData/user/kip/11 500 1.609 ms - 203
PatchConflictError: Array location out of bounds or not an instance property
GET /@mxtommy/kip/assets/svg/icons.svg 304 1.940 ms - -
GET /@mxtommy/kip/ 304 1.261 ms - -
GET /@mxtommy/kip/chunk-EAJKXTFC.js - - ms - -
GET /@mxtommy/kip/assets/steelseries-min.js 304 4.115 ms - -
GET /@mxtommy/kip/assets/NoSleep.min.js 304 3.386 ms - -
GET /@mxtommy/kip/chunk-ACMPUT6Z.js 304 2.647 ms - -
GET /@mxtommy/kip/assets/tween-min.js 304 6.354 ms - -
GET /@mxtommy/kip/chunk-OTD7HT32.js 304 2.086 ms - -
GET /@mxtommy/kip/chunk-AYIHSTCC.js 304 0.889 ms - -
GET /@mxtommy/kip/chunk-JZUG3W7P.js 304 5.820 ms - -
GET /@mxtommy/kip/chunk-SHKAX4XV.js 304 7.515 ms - -
GET /@mxtommy/kip/chunk-AB7BX4CL.js 304 5.110 ms - -
GET /@mxtommy/kip/chunk-MR7RUNKP.js 304 3.774 ms - -
GET /@mxtommy/kip/chunk-WE4P4WBD.js 304 2.923 ms - -
GET /@mxtommy/kip/chunk-36TH3DJS.js 304 2.117 ms - -
[... 119 lines cut ...]
GET /@mxtommy/kip/chunk-AB7BX4CL.js 304 2.740 ms - -
GET /@mxtommy/kip/chunk-MR7RUNKP.js 304 2.271 ms - -
GET /@mxtommy/kip/chunk-WE4P4WBD.js 304 1.734 ms - -
GET /@mxtommy/kip/chunk-SHKAX4XV.js 304 4.804 ms - -
GET /@mxtommy/kip/chunk-36TH3DJS.js 304 1.303 ms - -
GET /@mxtommy/kip/chunk-JZUG3W7P.js 304 4.134 ms - -
GET /@mxtommy/kip/chunk-QMIJ5LO6.js 304 4.885 ms - -
GET /@mxtommy/kip/polyfills-2DK6FBPF.js 304 3.494 ms - -
GET /@mxtommy/kip/media/KFOmCnqEu92Fr1Mu4mxKKTU1Kg-SNGEW7FX.woff2 304 2.015 ms - -
GET /@mxtommy/kip/styles-ILFVF6BG.css 304 1.449 ms - -
GET /@mxtommy/kip/main-QTKHCBQH.js 304 2.940 ms - -
GET /@mxtommy/kip/chunk-6QHCHGTT.js 304 5.848 ms - -
GET /@mxtommy/kip/chunk-CD375FDG.js 304 4.644 ms - -
GET /@mxtommy/kip/chunk-EQAHQINY.js 304 2.952 ms - -
GET /@mxtommy/kip/chunk-7TWI6ZJI.js 304 2.394 ms - -
GET /@mxtommy/kip/chunk-SGYCFQGT.js 304 3.885 ms - -
GET /@mxtommy/kip/chunk-EQ5CPWIU.js 304 1.929 ms - -
GET /@mxtommy/kip/chunk-Y7X77HSW.js 304 1.059 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOlCnqEu92Fr1MmSU5fChc4AMP6lbBP.woff2 304 4.268 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOlCnqEu92Fr1MmSU5fBBc4AMP6lQ.woff2 304 3.483 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOmCnqEu92Fr1Mu7GxKKTU1Kvnz.woff2 304 2.934 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOmCnqEu92Fr1Mu4mxKKTU1Kg.woff2 304 2.401 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOlCnqEu92Fr1MmEU9fChc4AMP6lbBP.woff2 304 1.916 ms - -
GET /@mxtommy/kip/assets/google-fonts/KFOlCnqEu92Fr1MmEU9fBBc4AMP6lQ.woff2 304 1.444 ms - -
```

gregw, 2025-05-26:
Hmmmm it is not infinite, as it eventually stops.  But the window never loads, so that could just be the browser throttling busy js?
Also, I've turned off all connections, so no data at all is getting in and it still is happening. So perhaps was coincidental that it first happened when I first tried my wind data?

godind, 2025-05-26:
The error is writing to the config file at:

POST /signalk/v1/applicationData/user/kip/11 500 1.875 ms - 203

PatchConflictError: Array location out of bounds or not an instance property

Not sure what the problem is but it's probably in the config file. Like corruption or something. Or an issue writing to that folder.

I'd first backup the config, clear it and see if it works.

gregw, 2025-05-28:
@godind I have files:
```
.signalk/applicationData/users/esprit/kip/11.0.0.json
.signalk/applicationData/users/kip/kip/11.0.0.json
.signalk/applicationData/users/admin/kip/11.0.0.json
```

All have valid looking content.  The user 'kip' no longer exists. I last used 'esprit' with kip and it has read/write permission?
Permissions look good on those files.

In the end I re-installed and then recovered a backup and everything is working fine now.  So perhaps close and I'll reopen if I break it again in a way that we can debug