mxtommy/Kip #649: A few presentation issues/questions

motamman, 2025-06-06:
Thanks, David, for all your work. With sailing season underway, I have a few thoughts. These are minor.

1. In the capacity gauge, could it have the option of a center number? (in most cases, probably 0) I have seen this in other gauge libraries. Consider this:

[image]

It would be nice if the negative numbers fell on one side of the gauge and the positive numbers on the other.

2. When using TTL in a widget, if the data times out, the widget should reflect a null state rather than a minimum state.

Consider these two items: One shows water below the keel in a numeric box, and the other shows actual depth using a linear gauge. (I start the actual depth gauge range at 7 feet, because that is what I draw. But that doesn't really matter. It could be zero.)  In the numeric box, when the data times out, I get dashes. The gauge, however, shows 7 feet. The point is that it displays a value. It should reflect a null, either blank or dashes.

[image]

3. When using a linear gauge with a needle indicator, the two needles are pointing to a spot that lacks any reference; it is centered over the imaginary area that would be filled if the widget weren't using a needle. Could the two needles be moved to center on the tick marks, or, perhaps better, have one needle in that fill area that points to the tick marks?

3b. The labels for text boxes don't conform to the same rules as labels on numeric boxes. They are sometimes tiny, tiny. I know this was a bit of a wrestling match at one point, so maybe you just made an editorial decision. Still, pointing it out,

[image]

Again, many thanks for all your hard work.

Pax,
Mo

godind, 2025-06-10:
Hi Mo,

Thanks for the feedback. Much appreciated.

1. I noticed and tried to change the position when I started with this gauge. It does not support positioning the labels or value, only font size and colour. We would need to use another gauge library and for all its fault, I can't find a better one that's free and open source. We could go for a commercial product but that would cost $. I can probably make the Capacity Gauge configurable with start position of: `0, 90, 180 or 270` degrees setting. Users will have to make sure the scale range they use matches the selected start position they to make it visually appealing: like with a range of -10 to +10. To have 0 in the upper middle it would be set to a start position of 180.
2. Good catch. I'll fix the values box when `null`.
3. I'll try and improve if the gauge library allows it.
3.b I'm not sure what text you are referring. Can you share a comparison screenshot?

motamman, 2025-06-10:
Regarding 3b. It isn't exactly as I had understood. The shrinking label affects both numbers and text boxes.
This is what it looks like when a number and text widget are added to a dashboard.

[image]

You can see how the labels look fine.

But, navigate away and come  back, and this is what the widgets look like:

[image]

I hope that helps.

godind, 2025-07-07:
No not dashes. Like you said just the lowest value.

The gauges, they are part of a library, should allow using a text to replace the valueBox value but it's not exposed in the Angular version so it's not possible. The library is not supported anymore and I can't find anything better that is Free!

The only thing I can think of is forcing zero even it's not part of and range.

motamman, 2025-07-07:
or go really old school and -999