mxtommy/Kip #662: Missing target heading in AP widget

arodus, 2025-06-22:
Hey, no matter what I'm trying the target heading is not being populated. The paths are setup correctly and the value is available in SignalK.
[image]

godind, 2025-06-22:
Hi. Can you share screenshots of your Signal K Data Browser page with the path and the KIP AP paths Configuration please?

Thanks

arodus, 2025-06-22:
[image]

[image]
Sure, there you go. Thank you!

arodus, 2025-06-22:
Amazing, thank you so much for the quick response.

godind, 2025-06-24:
I'll day a few days as I'm on a roadtrip and still have to fix a few things. I'll share a beta version you can easily install/uninstall.

searay24, 2025-07-03:
also waiting
looks and works great zones plugin config now to large signalk wont backup any longer

[image]

godind, 2025-07-06:
Will be included in next release.

closed 2025-07-06 by godind

godind, 2025-07-09:
Great feedback! Merci pour la rapidité du retour !  I'll work on improvements.

For the dial, it's like most AP head, is it not? The challenge is the locked heading/Bearing/wind angle does not change but for a couple of degrees so it worthless to use those value to rotate the Dial. We also need to think about using it as the only widget, say on a phone where comparing locked vs actual is valuable. If you have the AP, maybe you don't need to use another heading widget?

Are you on Discord? This issue is closed and so it's hard to track. I'm david_29482 or David Godin on the SK Discord server. You'll find me in the KIP channel.

godind, 2025-07-10:
We will keep using the current heading in the dial. See previous explanation. Also values on each side of the dial are part of the whole. All we would have if we remove the dial are buttons.

The form path issue is a long standing design issue. SK does not send paths until they have data. So paths are missing sometimes. Like next waypoint. KIP enforces path validation to close the form. 

It's a problem with the path selection component that affects more than the AP Head and that I have to look into for an alternative, but one that still forces the use of good paths and available ones only. Else 75% of the support request we get resolve to path issues.

What AP do you have? I suggest fixing you AP not sending target AWA instead.