mxtommy/Kip #725: Permission Denied accessing SignalK PlugIns

SirReal-surreal, 2025-07-25:
on one of my KIP dashboards I am using the embedded component to integrate a NodeRED dashboard. In this case the worldmap node.

This is so cool because I get a OpenSeaMap map on my KIP dashboard including my anchor, radius etc.

BUT: I need to turn off the SignalK security completely. 
Because I get a permission denied reply otherwise. 
Even though I am loged in via KIP, having a valid token. It would be great if the token can be added to the request also using the embedded widget because it seems that this is not the case currently.

godind, 2025-07-25:
What's being called and denied? I don't think OpenSeaMap is trying to talk to SK. I'd be surprised if knew how to.

SirReal-surreal, 2025-07-26:
The access is on NodeRED and the worldmap node:
http://ip:3000/plugins/signalk-node-red/redApsI/worldmap/

and I get this message:

[image]

I can do the same from my PC on KIP but having the chance to log in on SignalK directy before - getting what I expect:

[image]

godind, 2025-07-27:
Does it work when you're not embedding in KIP?

SirReal-surreal, 2025-07-27:
1) It works if I access it directly, yes, because I can log in in Signal K first. It also works if I turn off Signal K security completely, so PlugIns do not require that you log in in SignalK first.
2) The anchor alarm plugin is doing the calculation, but my solution also does a visualization, you see a circle and your ship's position within the circle, you can move the circle, you can extend (safety radius) or reduce it. You see your track (sway at anchor) and I have full access to all further improvements. Especially also regarding notifications: I can check if I still get positions via GPS or trigger a watchdog (not sure what the plugin does), I can deal with all notifications as I like.

I used the plugin first, just to have the calculation outsourced, however, compared what I did now this is the tiny part and just one little function node to calculate radius and create alarms, if necessary.

Long story short: The solutions is quite cool and it took some time but I am very happy. I will share this project because I think this can be a nice solutions for others as well. 
But this leads me to my access issue... I still can't access it w/o removing the security in general.

SirReal-surreal, 2025-07-29:
Yes, I am running KIP on a Raspberry 4/5 and on the same machine Signal K. Funny: On my PI4 it did work as I did a test, now working on my PI5 the issue is as described above.

I do nothing directly in SignalK but if you want to access any SignalK PlugIn, like NodeRED, you need to log in in SignalK first. My WorldMap Node is in Node-RED, so I do not get access without loggin in to Signal K first, if security is enabled.

BTW: I think this issue will also occur if you embedd a Node-RED Dashboard, which is widely used and known and would be great to integrate on the KIP embedded widget.

godind, 2025-07-29:
There is not much I can do as KIP does not use cookies. When you sign in to SK, it generates a cookie and the NodeRed plugin reads and uses. Look for a way to login to SK using nodeRed.