mxtommy/Kip #740: Time Widget freezes

eburi, 2025-08-01:
Hi,
I'm using KIP as instruments panel inside FreeboardSK. The dashboard I created has a Time Widget and then under it Grafana integrated.
The FreeboardSK is running nonstop on a TV screen (Samsung The Frame) with the instruments open. I noticed that the time widget stops working again and again. Then I need to close the instruments panel and reopen it to get it working again. 

I'm trying to reproduce it in Chrome on my MacBook, but so far I didn't have a chance and I'm well aware that it might be really hard to reproduce. I'm creating this issue to see if other people noticed the same behaviour. It might also be related to the browser integrated in the Samsung TV. It's also hard to tell when it actually stops. I only notice it ever so often when I walk by and see that the time is wrong.

Note: Integrated Grafana keeps working. No problem there.

If there is anyway I can help to debug this, let me know. On the Samsung TV I have not found a way to get to the debug console.

godind, 2025-08-01:
We need logs from the TV to see what's going on.

closed 2025-09-08 by godind

godind, 2025-09-09:
@eburi FYI it's not a Samsung fix but it still might be interesting to know: v3.10 has a built-in remote control feature enabling you to change the Samsung TV dashboard without using a keyboard/mouse.

I've also reduce GPU buffer allocation requirements considerably (graphics rendering optimizations). It might help the TV. Let me know if it improves things...

Cheers