mxtommy/Kip #747: Feature Request: Icons for switches

SirReal-surreal, 2025-08-05:
I played a lot with the KIP dashboard in the past days and it has some pretty nice functions.

One thing I am missing is a smaller switch and instead of text a simple icon, e.g. utilizing the free font-awesome library (like Node-RED does). Text is hard to read within the switches, especially because the non-pressed state of the box is always grey. Only text and pressed state can be colored. 

So, instead of defining a text it would be pretty cool to define a fa-icon and get this as a switch instead of the box. It would be great if you can define the color while not pressed and color while pressed. Only difference between button and switch: The switch stays in pressed color until it is pressed again.

These tiny switches would save a lot of space and would give a better indication. And, they would look much nicer as text in a box or next to a switch.

Just an idea, but I would love it 😀

godind, 2025-08-06:
Thanks for the feedback. I'll see if I can change the Colors around to improve visibility.

I've remove FontAwesome sometime ago to reduce dependencies and installation size in favour of built-in material icons and custom SVG icons.

It would be nice to have an icon set. The challenge is building tools to select, display and manage icons is a bit of work and will require maintenance down the road so we will need to weigh in benefits vs priories.

That said, keep sending feedback and ideas!

godind, 2025-08-07:
I'm not sure I fully understand what behaviour you are experiencing with not setting the path. Can you tell me more about the usecase?

Boolean path are either true or false. If you use the switch control it will change from one state to the other automatically. You can use the same indicator control if you just want to display the state or push bottom to repeatedly send true.

I hope this helps.