mxtommy/Kip #770: KIP feezes

jimmyendt, 2025-08-14:
i am sailing the last 2 weeks, have updated KIP everytime! and i love it

i run KIP on a rpi 4 with a sailor hat, SSD and cooling on an openplotter v4  install

on all versions in the last weeks KIP freezes sometimes.. like all the data just freezes.. the only way to get it online again is restart the raspberry pi ( or close chromium)

I remember there was an issue posted about this?

currently on KIP v 3.8.0 / chrome 139 

running on Signal K Server 2.14.4

jimmyendt, 2025-08-14:
i see there is an updste to server 2.15.1 

but Signal k keeps working and sending all the dsta to my ipad (for navigating) it is only my KIP screen on the rapberry oi that freezes

godind, 2025-08-15:
@jimmyendt  How much RAM do you have on the RPi and what OS version do you have installed?
Can you see any browser console log error?

Try running Firefox to see if you have the same issue

godind, 2025-08-15:
Framework memory leak patch from a library will be included in v3.8.0+. I don't think it is the cause of the issue at hand but it plugs a hole.

jimmyendt, 2025-08-15:
wow sounds good

thanks for all the work

i will wait untill new release and check witb chromium again

godind, 2025-08-21:
@jimmyendt @Techstyleuk @SirReal-surreal try out v3.8.1 on Chromium and report back please.

Thank you

jimmyendt, 2025-08-21:
wow that sounds like a lot of work! thank you so much!

i'm still sailing for 2 weeks so will test everyday:)

gregw, 2025-08-24:
That does sound at least like a different symptom if not a different root cause. If chrome is leaking memory, then the exact failure mode may vary. 

I've not seen any locks recently on my bench test setup, but I have on my boat (running older software). I'll update one display to use Firefox and see if there is a difference.

Techstyleuk, 2025-08-25:
I ran on Chromium for about 5hrs on Sunday, no freezes with version 3.8.1

godind, 2025-08-26:
I don't think iChromium is leaking but it does not handle GPU memory allocation the same way. The RPi Chromium build has less memory resources allocated for GPU, given the hardware, it's a RPi OS team decision I'd say. The Firefox build we install on RPI is probably built for desktop. The crash is worst on RPi Zero 2 it barely runs.

That half drawn broken wind widget is a symptom of GPU physical memory pressure. There app still runs fine. It's just the UI that does not update anymore (Kill Chromium renderer process)

If you scan your system log, you will find the OS killing a Chromium process (the renderer) because of memory pressure. Each canvas and SVG elements (what KIP uses to draw gauges) take up physical RAM that can't be swapped to file. Like from 200kb to 3mb each. Anyway...

I do think we have a leak somewhere but not sure where. I'm optimizing to reduce GPU memory requirements. It's healthy and I might run over the leak.

What would help is finding if widgets, and which one, is at fault, if any.

godind, 2025-08-26:
@SirReal-surreal 
> as I use it in kiosk mode for
It would be really nice if you could write/find instruction on how to setup KIP in kiosk mode. I could Add it ti the Help and GitHub Readme! and share the credit

jimmyendt, 2025-08-26:
i also use it in kiosk mode, but i made a script on startup in the rpi software. does that help?

godind, 2025-08-27:
> i also use it in kiosk mode, but i made a script on startup in the rpi software. does that help?
If it's reusable with parameters and such and you want to write about it sure! Do you know how to make GitHub PR? I can show you where and how to contribute and write a Help page and update  the landing project page.

It would also be nice to include an explanatory text. Use AI!

Techstyleuk, 2025-08-27:
@godind,  I use a 7" monitor in Kiosk mode and this is what I have in my notes as setup:

To get your Raspberry Pi to automatically start in Kiosk mode running Chromium browser, you'll need to set up an autostart script and configure your system to boot directly into the desktop environment. Here's how to do it:

Create a shell script for launching Chromium in kiosk mode:

`sudo nano /home/pi/kiosk.sh`

Add the following content to the file:

```
#!/bin/bash
xset s noblank
xset s off
xset -dpms

chromium-browser --noerrdialogs --kiosk --start-fullscreen --enable-features=OverlayScrollbar --app=http://<sk_server_IP>:3000/@mxtommy/kip/#/page/0
```

Save and exit the file (Ctrl+X, then Y, then Enter)

Make the script executable:

`sudo chmod +x /home/pi/kiosk.sh`

Configure the system to autologin to the desktop:

`sudo raspi-config`

Navigate to System Options > Boot / Auto Login and select Desktop Autologin12.

Create an autostart file:

```
mkdir -p ~/.config/autostart
nano ~/.config/autostart/kiosk.desktop
```

Add the following content:

```
[Desktop Entry]
Type=Application
Name=Kiosk
Exec=/home/pi/kiosk.sh
X-GNOME-Autostart-enabled=true
```

Save and exit the file

Reboot your Raspberry Pi:

`sudo reboot`

After rebooting, your Raspberry Pi should automatically start Chromium in kiosk mode, displaying the specified web page

Techstyleuk, 2025-09-02:
I can't remember if I changed dashboards, I don't think I did.  I was cruising and using the dashboard below.  I didn't have a Location set in OpenCPN so the VMG, Time to go,  distance to waypoint, Angle to waypoint, Cross track error and Heading to Waypoint were not populated.

[image]

I did find something on Sunday that the number of Deltas per second was between 1200-1500 and that about 1000 of those coming into SignalK, were coming in from OpenCPN, but OCPN should be limited to APB, RMB, XTE.

godind, 2025-09-04:
> don't think I did. I was cruising and using the dashboard below. I didn't have a Location set in OpenCPN so the VMG, Time to go, distance

I'm running a few tests but so far I can confirm KIP is not leaking memory, animation requests or generating code exceptions. I've reduce GPU buffer requirements to bare minimum, optimize canvas and SVG animations, triple checked no leaks are present. CPU utilization is stable at 40-60%. Chromium JVM heap utilization is acceptable at a stable 40-60% allocation.

Symptom:  Screen draw stops updating after a long period of time. App is still running fine but for the UI not updating. Diagnostic for far: Request Animation Frame (rAF) suddenly escalate infinitely when GPU/GL Context lost or corruption. The logs report following:
- ContextResult::kFatalFailure: TransferBuffer::Initialize() failed
- CommandBufferHelper::AllocateRingBuffer() failed

Things to investigate:
- Is caused by GPU driver bug ? Not sure yet but possible chromium --disable-gpu does not freeze.
- Is Chromium throttling KIP after a long period of inactivity and causing the crash? See --disable-background-timer-throttling --disable-renderer-backgrounding --disable-backgrounding-occluded-windows
- Is OS throttling KIP after a long period of inactivity and causing the crash? See fake mouse move script to simulate activity.

godind, 2025-09-07:
@gregw @jimmyendt @SirReal-surreal @Techstyleuk 

> ### Things to investigate
> 
> * Is caused by GPU driver bug ? Not sure yet but possible. chromium --disable-gpu does not freeze.

Ran for 2 days without crash with --disable-gpu option. My opinion is the bug is in the Chromium GPU driver. See bug report.

> * Is Chromium throttling KIP after a long period of inactivity and causing the crash? See --disable-background-timer-throttling --disable-renderer-backgrounding --disable-backgrounding-occluded-windows

Tested and has no effect on crash

> * Is OS going to Power Save mode and causing the crash? See fake mouse move script to simulate activity.

Tested and has no effect on crash

godind, 2025-09-07:
If you still want to use Chromium until the bug is fixed, disable GPU driver on Chromium launch. All processing will be using CPU. I've not seem perf degradation as KIP is not very GPU intensive. Use:
--disable-gpu

Other interesting Chromium option to use if running in kiosk mode are:
--disable-background-timer-throttling --disable-renderer-backgrounding --disable-backgrounding-occluded-windows

godind, 2025-09-07:
@Techstyleuk 
> [@godind](https://github.com/godind), I use a 7" monitor in Kiosk mode and this is what I have in my notes as setup:
> 

Excellent! I've added a Kiosk mode Help file. Have a look: https://github.com/mxtommy/Kip/blob/master/src/assets/help-docs/kiosk.md

SirReal-surreal, 2026-01-11:
> [@Techstyleuk](https://github.com/Techstyleuk) [@SirReal-surreal](https://github.com/SirReal-surreal) [@jimmyendt](https://github.com/jimmyendt) [@gregw](https://github.com/gregw) Has anyone had Chromium freezes recently?

Freezing cold winter here in Germany - so, no sailing currently, so I can't add experience currently, sorry.

Techstyleuk, 2026-01-12:
> [@Techstyleuk](https://github.com/Techstyleuk) [@SirReal-surreal](https://github.com/SirReal-surreal) [@jimmyendt](https://github.com/jimmyendt) [@gregw](https://github.com/gregw) Has anyone had Chromium freezes recently?

I am also out of the water, but at the end of the season I was not seeing this issue