mxtommy/Kip #790: Source for background graph in numeric widget

Asw1n, 2025-08-29:
The numeric widget can show a background graph.
In a situation where there are initially two sources (A and B) for the same path a numeric widget can be configured to use source B and to show a background graph. This all works fine. If now source B is no longer updated or (after a restart of signalK) is no longer available then thw widget will show -.- as its value. This is as expected. The background graph however still displays a valid value, one that is coming from source A. This should not be happening. The graph should not show valid values. 

I suspect that the background graph does not use the source settings of the widget.

godind, 2025-08-29:
Do you have Source Priority configured for this path?

Asw1n, 2025-08-30:
No, I have not.
As a matter of fact I am testing my own plugins (speed and current, advanced wind) that improve sensor data. I use KIP to compare input to output of my plugins. It was while testing that I noticed.

godind, 2025-08-30:
Here is what might be happening. When you initially create the widget you select the path. If it only have one source at that time, you can only select source Default.

Default reads from all sources for the that. If you add more Source or change Sources it will read everything mixed up together.

If you edit the widget and change Default to a specific Source, it should only use the source.

If you mix Sources dynamically, you need to configure SK Source priority rules. That way KIP reads from default all the time and SK handles sources its self.

godind, 2025-09-08:
@Asw1n did my suggestion resolve the issue?