mxtommy/Kip #816: AWA AND TWA on the race steer

petsicola, 2025-09-14:
Many people get confused when looking at ORC certificates and then trying to match those angles on their instruments or apps. Here’s the key point:

ORC certificates always use TWA (True Wind Angle). 
https://data.orc.org/public/WPub.dll/CC/035100034UV.pdf you can see here my certificate

On board, most sailing displays (like the mxtommyKIP wind rose) show both AWA (Apparent Wind Angle) and TWA.

If you take the “target angles” from ORC (e.g. ~40° upwind) and put them directly on the AWA needle, the boat will never match. Why? Because:

AWA is compressed by boat speed. Upwind it usually reads ~25–35°.

TWA is always wider — for the same situation that shows ~30° AWA, the TWA will be closer to 38–45°.

That’s exactly what happens: ORC says 40° (TWA), your instruments show ~30° AWA, and both are right.

How to fix it in your app / instruments:

Always compare ORC targets against TWA, not AWA.

If you want to give the helmsman an AWA target (e.g. “sail at 30° AWA”), you first need to convert the ORC TWA target into the corresponding AWA, using your current boat speed and wind.

A good display setup is:

Tactician / navigator: looks at TWA vs. ORC polar targets.

Helmsman: can steer to a stable AWA (simpler to hold), but knowing that AWA target comes from the polar’s TWA.

So the bottom line is: don’t mix up ORC TWA targets with AWA displays. If you do, the numbers will never line up.

Just to clarify with the example I posted earlier:

👉 In my screenshot I mentioned that my apparent wind angle (AWA) is usually around 30° upwind. That’s correct for the orange needle (A).

But the angles in the ORC certificate are always TWA (True Wind Angle), not AWA. On the mxtommyKIP wind rose that corresponds to the yellow needle (T).

So when the ORC polar shows ~40° upwind, that means 40° TWA → which in practice matches the ~30° AWA I usually see on board.

Bottom line: always compare ORC targets to the yellow TWA needle. If you try to match them against AWA, the boat will never line up because AWA is narrower than TWA due to boat speed.

[image]

[image]

petsicola, 2025-09-14:
As you can see in the screenshot, with my boat sailing at about 30° apparent wind angle, the heading lines up very closely with the ORC upwind target — so it feels like we’re right on the maximum pointing angle.

But the catch is: I’m looking at it using apparent wind (orange needle). On top of that, the target bars in the display are currently locked to AWA, when in fact they should be locked to the yellow T (TWA).

That’s why it looks right with 30° AWA, but in reality the ORC certificate angles are 40° TWA, and those targets should follow the yellow needle.

[image]

godind, 2025-09-16:
You're talking about the Winsteer, right?

The windsteer is a basic widget for sailors that do t need much. It only has a fix close haule angle that is used to draw the laylines and wind shifts. It uses as little Signal K path as possible.

I am building a new Racesteer widget that read data from the Polar Performance plugin - you need a  boat polar to use it. This one I plan on doing the kind of calculations you are explaining so the helmsman can see how far for best VMG steering angle he is, boats speed actual over polar potential, etc.

Are you talking about Windsteer or Racesteer?

Knowing that the Windsteer is basic and should stay simple, do you see a problem with AWA & TWA?

godind, 2025-09-16:
Oh I see. I made a mistake! The Racketeer you see is not the right widget. I must have published the very first Windsteer widget clone by mistake. Sorry about that. The Racesteer is just a Windsteer widget.

[image]

godind, 2025-09-18:
Good. It's how the Racesteer should work, with a bit more information.

The stream file you shared, was it just a sample to validate or was it a full race I can use?

I just looked quickly to see if it worked but did not give it more attention.

petsicola, 2025-09-18:
This is the file I have as of today with the data you requested. I have others, but I’m not able to get them working. If you could tell me how you’re reproducing them, I could provide you with examples of sailing close-hauled and downwind at full speed, and compare them with our polar charts.

petsicola, 2025-09-18:
I'm using this wind as  TWS. I don't have speed throug water. why? because the sensor is allways dirty and I Don't use it. in all my calculations I use as wind speed that path

[image]
And as you can see at the next image it gives values with speed through water... can we choose about our path? the path sog is self.environment.wind.speedOverGround

[image]

In this image if I put in drift less that 2000 ms it puts inconsistent data...  

[image]

[image]

[image]

I’m sending you a race. It wasn’t a good one, but you can see the other boats on AIS. I think you’ll notice some inconsistent data values on kip.. kip gives some values and data browser give another values.
https://we.tl/t-XcXYiZCLYP

petsicola, 2025-09-19:
Yes I have the sensor clean. The next time 2025/10/4 I have the next race and I'm going to put it and check the results and I send you the new log route But I think that you can use this data today.

godind, 2025-10-15:
Racesteer will have a rotating faceplate so that you can always see actual compass directions to visually guide you toward the next tack or waypoint.

With polars we can see what the best angle for VMG is. Instead of showing the number and having the skipper trying to hold that number, I plan to use the red and green tack lines indicators to show the best angle. On close-haul you should try and make that line parallel to the speed arrow (in the middle to the compass). The tack lines will narrow and enlarge depending on what the polar ways the best angle is at that moment.

Well... that's the concept anyway. With a good up wind, close-haules to close-hauled tacking recorded streams file, I'll be able to troubleshoot the math and wee what the polars says.