mxtommy/Kip #829: New basic Component wind information

hugorobi, 2025-09-30:
Can we have a simplified Wind Component that look like this one : 

[image]

Alose, you can add AWS and TWS on top left and top right as for the actual Wind Steer, and add option to display those values, or not.

godind, 2025-10-02:
With the current windsteer widget, you can turn off most of features. 

What else would you like to turn off? Is it that it adjust in relation to the heading?

hugorobi, 2025-10-02:
Don't want the compass and other "navigationnal" stuff, just wind information.

Having an option to show as "Compass" or as "Wind Direction" would be great (or another Component called Wind). "Compass" view change based on heading value, but "Wind Direction" is always the same because it is related to the boat.

For comparison : 

Compass option looks like this : 

[image]

And "Wind direction" look like this : 

[image]

You can keep TWS and AWS options as it is now.

It would also be great to have the option to display or not "Magnetic Heading" at the top center.

godind, 2025-10-15:
@petsicola You mean 180 on each side, right? I am of the opinion that a simple Wind widget makes sense with a fixed dial and just +/-180 because your looking at wind angle in relation to the boat - the apparent wind.

For a race widget it's different. You need bearings to next waypoint and such. It needs another design and to work a little differently.

@hugorobi I'll make a simple Apparent Wind widget, maybe options in windsteer... I have to think about it.

godind, 2026-01-20:
will be included in v4.3